atc1441/ZBS_Flasher #15: pins_arduino.h missing

maxholgasson, 2022-12-30:
Hi,
I'm new into platform.io and tried to upload ESP32_Flasher. However I'm running into this error
fatal error: pins_arduino.h: No such file..
Do I need to add that into the ZBS_Flasher folder?

atc1441, 2022-12-30:
Hey. 
That should not be needed.

It is enough to only open the ESP32_Flasher folder as a project in VSC Platformio

maxholgasson, 2022-12-30:
Mh ok, I tried it now a couple of times also with a fresh unpacked zip folder but still the same

> In file included from C:/Users/atem/.platformio/packages/framework-arduinoespressif32@3.20003.220626/libraries/SPI/src/SPI.cpp:22:
C:/Users/atem/.platformio/packages/framework-arduinoespressif32@3.20003.220626/libraries/SPI/src/SPI.h:25:10: fatal error: pins_arduino.h: No such file or directory

Is there a difference between open it as aproject or just the folder?

atc1441, 2022-12-30:
Please try to create a completely new esp32 arduino project in Plattform iO to see if that part works.
Something like a simple blink sketch

maxholgasson, 2022-12-30:
Ok,
seems there is something wrong in my setup when building it for the lolin32 lite when building it for the following it's builing and I could upload it
```
[env:esp32dev]
platform = espressif32
framework = arduino
board = esp32dev
monitor_speed = 115200
```