atc1441/ZBS_Flasher #16: Failed to read ZBS Flasher version

MeisterAdebar, 2022-12-30:
Hallo,

kannst du mir beantworten was der Grund für diese Meldung beim flashen der Custom-Firmware sein könnte.
Die Verkabelung hab ich mehrfach geprüft und ausprobiert hab ich es unter Windows mit der .exe und unter Ubuntu mit dem Python-Skript mit dem gleichen Ergebnis. 

```
PS C:\Users\bla\Desktop\ZBS_Flasher-main\ZBS_Flasher-main> .\zbs_flasher.exe
Manual: COM1 read/readI file.bin, or COM1 write/writeI file.bin 0 or slow_spi baudrate(default 115200) 
Example: COM1 read file.bin slow_spi 115200 <- will read flash to file.bin with slow SPI and 115200 baud
Example: COM1 write file.bin <- will write file.bin to flash with fast SPI and default 115200 baud
Example: COM1 MAC <- will write the original MAC into the infopage
Example: COM1 MAC 1234567890ABCDEF <- will write this 16 byte MAC into the infopage
Not the right arguments but here are the... please wait...
possible UART ports: COM5
```

```
PS C:\Users\bla\Desktop\ZBS_Flasher-main\ZBS_Flasher-main> .\zbs_flasher.exe COM5 write custom-firmware/demo_firmware_2.9_33_V1.9.0.0.bin
Using port: COM5
Failed to read ZBS Flasher version
Some Error in init
```

Gruß

Thomas

atc1441, 2022-12-30:
Es sieht so aus als wäre die Verbindung zum esp32 noch nicht einmal erfolgreich.

Sonst müsste er die Version lesen diese kommt vom ESP

Bitte versuche den esp32 Flasher erneut auf den esp32 zu flashen

MeisterAdebar, 2022-12-30:
> Bitte versuche den esp32 Flasher erneut auf den esp32 zu flashen

Hab ich vor meiner Frage mehrfach, jetzt nochmal, das flashen passt:

```
Processing nodemcu (platform: espressif8266; board: nodemcuv2; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 (4.0.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 1.200.0 (2.0) 
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 35 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0
Building in release mode
Retrieving maximum program size .pio\build\nodemcu\firmware.elf
Checking size .pio\build\nodemcu\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  36.3% (used 29712 bytes from 81920 bytes)
Flash: [===       ]  25.9% (used 270837 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM5
Uploading .pio\build\nodemcu\firmware.bin
esptool.py v3.0
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 24:d7:eb:c6:e3:8a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Compressed 274992 bytes to 201328...
Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 274992 bytes (201328 compressed) at 0x00000000 in 3.2 seconds (effective 689.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
============================================================== [SUCCESS] Took 10.91 seconds ==============================================================

Environment    Status    Duration
-------------  --------  ------------
nodemcu        SUCCESS   00:00:10.910
============================================================== 1 succeeded in 00:00:10.910 ============================================================== 
 *  Das Terminal wird von Aufgaben wiederverwendet, drücken Sie zum Schließen eine beliebige Taste. 
```

Aufgrund deiner Vermutung hab ich einfach mal alles vom ESP abgestöpselt, um meine Verdrahtung auszuschliessen und den Befehl mit dem "nackten" **Wemos D1 Mini** ausgeführt, dabei kommt es zur selben Meldung.

Dann scheint doch irgendwas am Sketch nicht zu passen ?

MeisterAdebar, 2022-12-30:
Eine Frage am Rande.

Die neuen Preisschilder von Lidl werden unterstützt ? Die großen für "oben" haben ja DIN A4 Größe  :o

atc1441, 2022-12-30:
> Eine Frage am Rande.
> 
> Die neuen Preisschilder von Lidl werden unterstützt ? Die großen für "oben" haben ja DIN A4 Größe :o

Nein

atc1441, 2022-12-30:
Ich sehe du benutzt einen esp8266 und nicht Esp32
Versuche dann mal bitte diese version
https://github.com/atc1441/ZBS_Flasher/tree/4a48b535e622f0d42181c1c4880cf19a5d9effeb

MeisterAdebar, 2022-12-30:
Vielen Dank, hilft aber nix, weiterhin : 
```
Failed to read ZBS Flasher version
Some Error in init
```

In der platformio.ini siehts so aus:
```
[env:nodemcu]
platform = espressif8266
board = d1_mini_lite
framework = arduino
upload_port = COM5
upload_speed = 921600
monitor_port = COM5
monitor_speed = 115200
```

atc1441, 2022-12-30:
Es sieht für mich so aus als würdest du das falsche projekt builden.

ist in der leiste unten "ESP32_Flasher" ausgewählt im Visual Studio Code?

atc1441, 2022-12-30:
Ich habe in der Aktuellsten Version noch etwas geändert, vorher hätte es theoretisch nicht für den ESP8266 builden können.
Das Serial1 "Pass Through" hätte eigentlich beim kompilieren meckern müssen.

MeisterAdebar, 2022-12-30:
Nee, keine Änderung, es werden beide Projekte kompiliert, ich wähle das richtige aus und es bleibt bei dem Fehler.

MeisterAdebar, 2022-12-30:
Halt Stop. Hatte **mit** der Verdrahtung getestet. Jetzt mal wieder alles abgestöpselt, "nackt" bekomm ich jetzt eine Version  :)

```
Using port: COM5
ZBS Flasher version: 32
Some Error in init
```

Jetzt such ich mal den Fehler bei mir  ::), kann dauern, melde mich.

Ganz großes Danke für die Hilfe und

Gutes Neues Jahr wünsch ich

atc1441, 2022-12-30:
Das sieht doch sehr gut aus.
Dann kommt als nächsten die Verbindung zum ZBS243.

Viel Erfolg soweit :)

MeisterAdebar, 2022-12-31:
Hab da offensichtlich noch ein Verständnisproblem  ::)

SPI_Clk -> 14
SPI_MoSi -> 13
SPI_MiSo -> 12
SPI_SS -> 15
RESET -> 0
GND -> GND
VCC verbinde ich **direkt** mit 3,3 V des ESP !

Wenn ich den Wemos mit Spannung versorge, dann versucht das Display sich zu aktualisieren, zu welchem Zeitpunkt muss ich die Spannungszufuhr unterbrechen und wann genau führe ich das _.\zbs_flasher.exe COM5 write custom-firmware/demo_firmware_2.9_33_V1.9.0.0.bin_ aus ?

atc1441, 2022-12-31:
Die spannungs versorgung muss erst nach dem Flashen unterbrochen werden.

In den Programiermodus kommt der ZBS243 über den Reset pin nur raus geht es nur über ein Repower

MeisterAdebar, 2022-12-31:
> In den Programiermodus kommt der ZBS243 über den Reset pin ...

Heißt, ich verbinde einfach nur mit dem USB-Anschluss, warte bis das Display fertig mit seiner "Aktualisierungsanfrage" fertig ist und führe dann den Flashbefehl aus ?
Genau das klappt nämlich nicht, dann bekomm ich weiterhin einfach nur den Initialisierungsfehler, ich bin mir eigentlich sehr sicher das alles korrekt verdrahtet ist, auch keine Brücken vorhanden (ich löte, hab keinen Jig).

```
PS C:\Users\bla\Downloads\ZBS_Flasher-main\ZBS_Flasher-main> .\zbs_flasher.exe COM5 write custom-firmware/demo_firmware_2.9_33_V1.9.0.0.bin
Using port: COM5
ZBS Flasher version: 32
Some Error in init
```

atc1441, 2022-12-31:
Vor dem flashen muss nicht extra gewartet werden, das geht direkt nach den anschließen.

Soweit sollte es eigentlich funktionieren und könnte soweit nurnoch an der verdrahtung liegen.

Du könntest noch versuchen den zbs_flasher mit "slow_spi" am ende als extra argument zu nutzen, vielleicht wird es dann etwas.

MeisterAdebar, 2022-12-31:
Hab einfach mal eine Pause gemacht, zurück, alle Dupont-Kabel an dem Wemos entfernt, neu angeklemmt, einen anderen USB-Port verwendet und weiß der Kuckuck warum der Flash dann geklappt hat:

```
PS C:\Users\bla\Downloads\ZBS_Flasher-main\ZBS_Flasher-main> .\zbs_flasher.exe COM3 write custom-firmware/demo_firmware_2.9_33_V1.9.0.0.bin
Using port: COM3
ZBS Flasher version: 32
Erasing flash now
Flashing file: custom-firmware/demo_firmware_2.9_33_V1.9.0.0.bin
File size : 32982
32982 /   32982  100% 8 seconds
Writing done, it took 8 seconds
Verfiy done and OK
ZBS Reset
Pass Through mode:

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00043260
~ld

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v00043260
~ld
```

Danke trotzdem für die Unterstützung.

Weiter gehts  ;D

MeisterAdebar, 2023-01-01:
Hallo und Frohes Neues!

Muss nochmal, was ich nicht erwartet hätte.
Das flashen der Custom-Firmware hat ja geklappt, auch das schreiben einer eigenen AC-Adresse.
Beim CC2532 war/bin ich der Meinung das der Flash auch geklappt hat, hat zwar 256 Sekunden gedauert, aber am Ende stand in dem zbs_flasher Programm ganz unten "Wrote ...". (das hab ich jetzt auch schon mehrfach wiederholt)

Wenn ich jetzt station.py ausführe wird der Stick auch erkannt, das Python-Skript gibt
`2023-01-01 14:45:47,229 Station started`
aus, mehr passiert aber nicht, ich sehe keine "Anfragen" wenn ich das Dispay boote, es passiert einfach nichts.

Die einzigen Einstellungen die ich vorgenommen habe sind diese:

```
CHECKIN_DELAY = int(os.environ.get("EPS_CHECKIN_DELAY", default="**3**00000")) # 900s
RETRY_DELAY = int(os.environ.get("EPS_RETRY_DELAY", default="1000")) # 1s
FAILED_CHECKINS_TILL_BLANK = int(os.environ.get("EPS_FAILED_CHECKINS_TILL_BLANK", default="**0**"))
FAILED_CHECKINS_TILL_DISSOC = int(os.environ.get("EPS_FAILED_CHECKINS_TILL_DISSOC", default="2"))
```

Wie geh ich jetzt weiter vor, wie finde ich heraus das das Display und der Stick auch wirklich korrekt funktionieren sollten ?

atc1441, 2023-01-01:
Die stick LED sollte soweit garnicht leuchten.
Denke da ist was beim flashen falsch gelaufen, du flash mit dem CC.Flash Programm? der ZBS_Flasher kann den CC2531 nicht flashen

atc1441, 2023-01-01:
Das ist jedenfalls zu lange ja.

was ich schon ein paar mal als Rückmeldung bekommen habe ist das es mit ausgeschaltetem verify funktioniert.
Am besten so ein paar mal Flashen.

MeisterAdebar, 2023-01-01:
OK, mach mich nochmal ran, zur Sicherheit: ich hab den Stick bisher immer beim flashen an einen USB-Port gesteckt das er mit Spannung versorgt wird, das will ich jetzt mal vermeiden, verstehe es doch richtig das dann _Target Voltage (2)_ und  3,3V (9) einfach gebrückt werden ?

atc1441, 2023-01-01:
Über USB versorgen sollte ohne Probleme funktionieren,
ansonsten kann aber 2 und 9 einfach gebrückt werden

Gehst du nach diesem Video vor ? https://youtu.be/98fOzZs__fc

MeisterAdebar, 2023-01-01:
> Gehst du nach diesem Video vor ?

Ja

MeisterAdebar, 2023-01-01:
> Das ist jedenfalls zu lange ja.

Es ist und bleibt aber so, das flashen dauert immer 276 Sekunden (nicht 256), dabei steht unten abwechselnd _Sending WRITE_XDATA ..._ und _Sending READ_CODE..._

Vorallem ist es ein Kraus das überhaupt mal ein Connect zustande kommt, ich ändere nix an der Verkabelung, selten klappts direkt.

Nach dem _Wrote Flash ..._ wird der Stick dann immer direkt erkannt, die LED wird grün (während dem flashen ist sie aus) und wechselt nach etwa 30 Sekunden auf rot.

[image]

MeisterAdebar, 2023-01-01:
Hab jetzt mal nach dieser Anleitung geflasht:

https://www.zigbee2mqtt.io/guide/adapters/flashing/alternative_flashing_methods.html

```
bla@bla-Aspire-A315-23:~/Downloads/CCLoader-master/SourceCode/Linux$ ./CCLoader /dev/ttyUSB0 firmware_TiMAC_cc2531.bin 1
Comport open:
Device: Leonardo

Baud:115200 data:8 parity:none stopbit:1 DTR:on RTS:off
File open success!
Warning: file size isn't the integer multiples of 512, last bytes will be set to 0xFF
Block total: 145
Enable transmission...
Request sent already! Waiting for respond...
Begin programming...
1  2  3  4  5  6  7  8  9  10  11  12  13  14  15  16  17  18  19  20  21  22  23  24  25  26  27  28  29  30  31  32  33  34  35  36  37  38  39  40  41  42  43  44  45  46  47  48  49  50  51  52  53  54  55  56  57  58  59  60  61  62  63  64  65  66  67  68  69  70  71  72  73  74  75  76  77  78  79  80  81  82  83  84  85  86  87  88  89  90  91  92  93  94  95  96  97  98  99  100  101  102  103  104  105  106  107  108  109  110  111  112  113  114  115  116  117  118  119  120  121  122  123  124  125  126  127  128  129  130  131  132  133  134  135  136  137  138  139  140  141  142  143  144  145  Program successfully!
File closed!
Comport closed!
bla@bla-Aspire-A315-23:~/Downloads/CCLoader-master/SourceCode/Linux$
``` 

Gleiches Verhalten, erst ist die LED grün und nach etwa 30 Sekunden wird auf rot gewechselt.

atc1441, 2023-01-01:
Das ist weiterhin sehr merkwürdig. 
in der Firmware werden die leds garnicht angesteuert, da muss also etwas falsch laufen

MeisterAdebar, 2023-01-02:
Mit meinen geschilderten Problemchen schein ich nicht alleine zu sein, vlt. liest du ja mal [hier](https://forum.fhem.de/index.php/topic,131303.msg1255041.html#msg1255041) mit und kannst uns mit Rat und Tat zur Seite stehen.

atc1441, 2023-01-02:
Gerade einmal durchgelesen, leider fällt mir da nicht viel zu ein.

Denke aber das es auch mit an der VM liegen kann. "Echtes" Windows könnte mehr Erfolg bringen.

Ansonsten vielleicht einen CC-Debugger nutzen, https://www.ebay.de/itm/124981579687

MeisterAdebar, 2023-01-05:
Moin, passt jetzt. Aber die Reichweite mit rd. 5 Metern ist nicht so berauschend. Kann man das Funknetz irgendwie erweitern ? Oder ist es technisch irgendwie möglich die Firmware des CC2531 so zu erweitern das ein Router-Device aus dem "normalen" Zigbeenetz als Router fungiert ?

atc1441, 2023-01-05:
Bei den china CC2531 modulen schwankt die reichweite leider.
Da hier nicht Zigbee selber genutzt wird sondern ein eigenes Protokoll auf dem (IEEE-802.15.4) Funkstandart basierend kann man leider kein anderes Zigbee gerät nutzen.

Ich habe mitlerweile sehr gute reichweiten erfahrungen mit dem CC2531 mit externer Antenne gemacht z.b. https://www.ebay.de/itm/372670770001