atc1441/ZBS_Flasher #17: zbs_flasher.py throws an error

roobbb, 2023-01-03:
Hello there.

At first many thanks for your great work and sharing your docs and sources 😀 😎
I saw your video on Youtube demonstrating your price tag hack - very impressive and gets me tinkering with such a stuff as shown.

I tried to flash a ZBS display by zbs_flasher.py. Just started without any parameters. But it failed with an error message:
```
sudo chmod +x zbs_flasher.py
./zbs_flasher.py 
  File "./zbs_flasher.py", line 312
    str(int((position/0x10000)*100)) + "% " + str(int((millis() - reading_start_time) / 1000)) + " seconds", end='\r', flush=True)
                                                                                                                ^
SyntaxError: invalid syntax
```

I even tried zbs_flasher.exe, but it seems this file doesn't run on 32Bit Windows. It runs fine on Win10 64Bit.

Did I do something wrong or is there anything missing?

Thank you and kind regards
roobbb

atc1441, 2023-01-03:
Hey. 
Could it be that linux does not support the ' braccets ? 
Please replace them with " at the error position

hixhix, 2023-01-03:
Hi.

This error looks like to be the result of using Python v2.
Please use Python v3

roobbb, 2023-01-03:
Thank you for the super fast response. I wasn't ware to be using any specific python version. Just started the script 😉
Now, I tried to force using python3 by changing the shebang line as follows:
`#!/usr/bin/python3`

The error message above doesn't appear but instead of that new errors show up:
```
./zbs_flasher.py 
./zbs_flasher.py: Zeile 5: import: Befehl nicht gefunden
from: Zu viele Argumente
./zbs_flasher.py: Zeile 7: import: Befehl nicht gefunden
./zbs_flasher.py: Zeile 8: import: Befehl nicht gefunden
./zbs_flasher.py: Zeile 9: import: Befehl nicht gefunden
./zbs_flasher.py: Zeile 11: CMD_GET_VERSION: Befehl nicht gefunden
./zbs_flasher.py: Zeile 12: CMD_RESET_ESP: Befehl nicht gefunden
./zbs_flasher.py: Zeile 13: CMD_ZBS_BEGIN: Befehl nicht gefunden
./zbs_flasher.py: Zeile 14: CMD_RESET_ZBS: Befehl nicht gefunden
./zbs_flasher.py: Zeile 15: CMD_SELECT_PAGE: Befehl nicht gefunden
./zbs_flasher.py: Zeile 16: CMD_SET_POWER: Befehl nicht gefunden
./zbs_flasher.py: Zeile 17: CMD_READ_RAM: Befehl nicht gefunden
./zbs_flasher.py: Zeile 18: CMD_WRITE_RAM: Befehl nicht gefunden
./zbs_flasher.py: Zeile 19: CMD_READ_FLASH: Befehl nicht gefunden
./zbs_flasher.py: Zeile 20: CMD_WRITE_FLASH: Befehl nicht gefunden
./zbs_flasher.py: Zeile 21: CMD_READ_SFR: Befehl nicht gefunden
./zbs_flasher.py: Zeile 22: CMD_WRITE_SFR: Befehl nicht gefunden
./zbs_flasher.py: Zeile 23: CMD_ERASE_FLASH: Befehl nicht gefunden
./zbs_flasher.py: Zeile 24: CMD_ERASE_INFOBLOCK: Befehl nicht gefunden
./zbs_flasher.py: Zeile 25: CMD_SAVE_MAC_FROM_FW: Befehl nicht gefunden
./zbs_flasher.py: Zeile 26: CMD_PASS_THROUGH: Befehl nicht gefunden
./zbs_flasher.py: Zeile 28: Syntaxfehler beim unerwarteten Wort »(«
./zbs_flasher.py: Zeile 28: `def print_arg_manual():'
```

So I tried another way:
`python3 zbs_flasher.py`

What throws:
```
Traceback (most recent call last):
  File "zbs_flasher.py", line 9, in <module>
    import serial.tools.list_ports
ModuleNotFoundError: No module named 'serial'
```
OK, it was already stated on README.md we need serial installed. So "pip install pyserial" didn't do the trick yet. Therefore I installed pip3 and via pip3 the needed serial:
```
sudo apt install python3-pip
pip3 install pyserial
```
After that it seems to be working:
```
python3 zbs_flasher.py 
Manual: COM1 read/readI file.bin, or COM1 write/writeI file.bin 0 or slow_spi baudrate(default 115200) pass at the end for UART Pass Through mode
Example: COM1 read file.bin slow_spi 115200 <- will read flash to file.bin with slow SPI and 115200 baud
Example: COM1 write file.bin <- will write file.bin to flash with fast SPI and default 115200 baud
Example: COM1 MAC <- will write the original MAC into the infopage
Example: COM1 MAC 1234567890ABCDEF <- will write this 16 byte MAC into the infopage
Not the right arguments but here are the... please wait...
possible UART ports: /dev/ttyS4 /dev/ttyACM2 /dev/ttyACM1 /dev/ttyACM0 
```

Thank you and kind regards
roobbb

atc1441, 2023-01-03:
Great. Thank you for that extended response as well

closed 2023-01-03 by roobbb