atc1441/ZBS_Flasher #25: Firmware Image with Grayscale

jhbruhn, 2023-02-09:
I am currently using the precompiled firmware image you provide here https://github.com/atc1441/ZBS_Flasher/tree/main/custom-firmware/, which as far as I understand was compiled from the code in `firmware_ch11_low_power` for a v033 ESL. 

Unfortunately I don't yet understand the changes you made to disable the grayscale colors of the epaper display. There are two board variants, `zbs29v033` and `zbs29v033_BW`, but both have the same LUTs, probably only capable of showing BWR images without grayscale?

My question is: What do I have to do to get the other benefits of this firmware, but also be able to show grayscale values?

jhbruhn, 2023-02-09:
Thanks. Were there any other changes required for the v033 board revision? Dmitrys version only has configs for v025 and v026.

atc1441, 2023-02-09:
It should be directly compatible with the v26 if i remember right

jhbruhn, 2023-02-09:
Judging by this, it should be the v26 https://github.com/atc1441/ZBS_Flasher/blob/0e2d9852bc659d38c6136b0abb2517855143651a/custom-firmware/firmware_ch11_low_power/board/zbs29v033/board.h#L43 
For future reference.

closed 2023-02-09 by jhbruhn

atc1441, 2023-02-09:
Yes it was Version 26 👍