atc1441/ZBS_Flasher #33: zbs_flasher.py can only read?

Smoeller0, 2023-06-22:
Hello
i just try to flash my first ZBS. The esp32 works the led flashes. 
I think i can read from ZBS chip but can not write? What can the possible error?
i also think the flash is erased. after this the chip uses 7mA at 3,3v and no more update the display.

python3 zbs_flasher.py -p /dev/ttyUSB0 read-infopage infopage reset
Welcome to the ZBS-Flasher Utility
Using port /dev/ttyUSB0
All done!

python3 zbs_flasher.py -p /dev/ttyUSB0 write demo_firmware_4.2_33_V1.9.0.0.bin reset
Welcome to the ZBS-Flasher Utility
Using port /dev/ttyUSB0
Erasing flash... Done.
Error: Writing to flash failed at 0x00000 tried to write 250 bytes: Writing command returned an error

atc1441, 2023-06-22:
Hey, what is the content of your dumped infopage? 

It could be a wrong wiring so reading is without errors but still wrong content

Smoeller0, 2023-06-22:
[image]
this i read from infopage.
what shuld be in the infopage dump? 

wrong wire perhaps. but when i remove any wire i become an read error. exept the txd rxd. are they used ? when disconect txd rxd read works also

atc1441, 2023-06-22:
Ok that dump does not loon correct, only the first 16bytes should be random but everything else should be 0xFF

Please check the wiring, the rx and tx is not needed for flashing.

Would suspect the power not being correct.
Please try to directly connect that to 3.3V

Smoeller0, 2023-06-22:
[image]
I get an new chip. i add an extra 100nf cap betwen vcc and ground. 
dump the original firmware. this looks like this. can this be right?
after first flash with same error the display notmore refresh. 

The esp32 sometimes need an reboot after read or try flashing. perhaps the error?
the message is timeout and no led on the esp32.

i use an 3,3v power supply direct on the vcc / ground. without reset schematic. 

so tomorrow i add an logicanalyser to see whats going on there.

atc1441, 2023-06-22:
The dump looks not too bad, but could be the same as with the infopage and only some bytes swaped.

Maybe try the low speed option

Smoeller0, 2023-06-23:
I try the --slow but the same problem still alive.
this is try reading. Miso looks like the first block is ok then its dying. 
[image]

writing data only one block then it stops. 
[image]

i re wire an nes esp32 a new display and have the same problem.

atc1441, 2023-06-23:
Could you please share a picture of your setup, like an overview and one on the solder job. 
Not meant bad just to find a possible solution :)

Smoeller0, 2023-06-23:
I found the problem. U where right. The Power was the Problem. My cable vom the power supply broke at the clamp but not complete. broke only nearly. measuring show 3.3v but short out only ~20mA then power broke down. 

Now it work fine. I ll thank you.

atc1441, 2023-06-23:
Perfect :)

closed 2023-06-25 by Smoeller0