atc1441/ZBS_Flasher #35: Error: timeout for ESP32 and ESP8266, any USB port

chunkysteveo, 2023-07-07:
Hi Aaron,

Trying to run the flashing program on a Windows 11 machine to either an ESP8266 and an ESP32 (wemos mini). Hooked up a tag to the ESP and all attempts to flash using "zbs_flasher.exe -p COM9 mac write Tag_FW_2.9.bin" result in Error: timeout.

Tried the --slow command with no luck.

Do you have any pointers on this to troubleshoot?

Thanks!

shin10, 2024-06-15:
Same here with a D1 mini ESP32.
My last lolin is currently broken. I'll try to fix it and check whether it's a problem with the board.

shin10, 2024-06-18:
OK, it had been the wiring. D1 mini ESP32 works just fine.

@chunkysteveo, is your tag refreshing the display, when you put it in the flasher? If not check VCC and GND, otherwise test the other ones. At least I had the esp detecting a device, even though the tag wasn't powered. All other problems I had where due to poor spacing of my pogo pins. They're not fitting perfectly in my print so I had to align them with some scraps of paper until it eventually worked; finally glued them with the paper in place. 😅

closed 2024-06-18 by chunkysteveo