atc1441/ZBS_Flasher #38: CC2652 UART pin layout

Smurfy37, 2024-02-25:
Hi tell me if im stupid i cant figure out the pin layout for flashing uart on my SONOFF zigbee adapter for a Epaper Link station as i have
3.3v
GND
TMS
TCK
Reset

Ive figured out 3.3 is VCC gnd is ground reset is RTS but upon research ive found TMS is a mode selection pin which doesnt sound like a RX or TX pin the TCK pin is a test clock pin which also doesnt sound like UART either please tell me if im stupid but im dying to get this working.
Thanks.
I really appreciate the hard work on this project if i could contribute some code to this project i would i think its fantastic keeping these devices from going to a landfill.

atc1441, 2024-02-25:
You can flash that SONOFF adapter directly via the USB connection, you need to hold the Boot button while reseting.

You need to search for a CC2652 flasher tool, this is not included in this repo

Smurfy37, 2024-02-25:
That's f***ing awesome I'm gonna set up the e paper link station now and
see what happens I appreciate your help I'll donate some money because you
deserve it

Smurfy37, 2024-02-25:
Ive got the recomended smart RF flash programmer 2 from TI and i get this error in the status while attempting to reprogram?
>Unknown record type: 3.
has this worked?
also im stuggling to find the open E paper link station executable is this something i need to compile? im new to all this stuff so im getting really confused.

LOGS
>Initiate access to target: COM8 using 2-pin cJTAG.
>Reading file: C:/.../EPaperStation_CC_Driver.hex.
>Unknown record type: 3.
>Reset target ...
>Reset of target successful.