atc1441/ZBS_Flasher #40: Timeout when flashing with ESP-Flasher

JohnnyMoonlight, 2024-12-30:
Hey, thanks for this repo. I have trouble flashing the ZBS Chip with the python script and receive a timeout-error.

I have built the ESP-based flashing interface with the ESP-flasher provided in the ESP32-folder. This is my platform.ini:

```
[env:nodemcu]
platform = espressif32@5.0.0
board = lolin32_lite
framework = arduino
upload_port = /dev/ttyUSB0
upload_speed = 921600
monitor_port = /dev/ttyUSB0	
monitor_speed = 115200
build_flags = 
#Changed to my esp32wiring	
	-D ZBS_CLK=18
	-D ZBS_MoSi=5
	-D ZBS_MiSo=17
        -D ZBS_SS=23 
	-D ZBS_Reset=19
	# do not connect directly to a GPIO only trough some kind of Mosfet or switch! **-> I use a separate ESP32 for powering up switched through a mosfet controlled by the flashing ESP32.**
  	-D ZBS_POWER=4 # D2 
	# Used to read UART data from the firmware running on the ZBS, not needed at all **-> I think this is related to my problem.**
        -D ZBS_RXD=16 -> changed to my board wiring.
  	-D ZBS_TXD=4 -> changed to my board wiring.
        -D LED=2
	-D ZBS_SPI_BUS=VSPI 
```

Now, running 
`$ python3 zbs_flasher.py -p /dev/ttyUSB0 --slow read ../SOLUM_29_SSD1619-ap-01-0017.bin`
gives this output:

```
Welcome to the ZBS-Flasher Utility
Using port /dev/ttyUSB0
Error: timeout
```
After sending the command, the zbs_flasher.py script awaits a UART response, as I understand it, and tries to find some specific content (zbs_flasher.py:65):
```
        self._serial.read_until(b"AT")
        buf = self._serial.read(2)
        if len(buf) != 2:
            raise CommunicationError("timeout")
``` 
The response does not contain the expected string, what am I missing? I thoght UART was optional according to the descriptions, yet the python script seems to require some UART-type of communciation. Any help is appreciated!

atc1441, 2024-12-30:
Hey, since the communication from python to the ESP32 is via UART this is where it is needed, seems like there is an issue in your upload or com port selection if no communication starts in between them

JohnnyMoonlight, 2024-12-30:
Hey, thanks for the quick reply and the explanation! It seems, my problem was  the `--slow` parameter, I used, as I had other problems before. Without it, I was able to communicate with the ZBS chip. Thanks again and have a good 2025.

atc1441, 2024-12-30:
Perfect :)
And good 2025 for you as well

closed 2024-12-30 by atc1441